MirageC79/Interfaces-for-WobbleX-integration #1: Rat-Rig Vcore-3

rabby7, 2023-03-31:
I guess it is not the same for Vcore-3 as Vcore-3.1
If you can make for Vcore-3 as well, I will be happy to get one set. Looks geat.

MirageC79, 2023-03-31:
I am trying to educate myself if both are equivalent.  Will get back to your shortly.

hapklaar, 2023-03-31:
First of all thanks for your work @MirageC79 ! Very excited to try this on my RR vcore 3.1

Curious to know if and how they differ regarding the arms as well, following!

Also can you specify how to print the various parts (orientation/supports)?

okpail, 2023-04-22:
Thanks for the work, the step files do not seem to be in utf8 format. Cannot open the files in fusion360... But maybe I am doing something wrong?

MirageC79, 2023-04-27:
Thank you, as shown by Teaching Tech in his video, V3 and V3.1 are not fully compatible.  I might not be able to create the adapter parts in a timely manner for V3, I hope the community can chip in.
The parts for RatRig V3.1 can be printed as oriented in the STLs with no support.  (Same as original parts)

closed 2023-04-27 by MirageC79

psiberfunk, 2023-05-21:
@MirageC79 im still interested in the vcore 3 (not 3.1 ) as well .. would it be possible to adapt still ?