MirageC79/Interfaces-for-WobbleX-integration #2: Print files of WobbleX

pasha4ur, 2023-03-31:
Hello.

Can you give the community (whose idea you used in this product ) the self-printing files (step and stl) for WobbleX?

Or are you afraid that you won't make all the money in the world by selling a device you created based on a free and open idea?

MirageC79, 2023-03-31:
1 - Credits were given to the original inventor from the start (who is part of this project by the way...)
2 - And CADs where published 3 month ago for a printable version here https://a360.co/40QC5G5

What are your motivations to go that hard on this? We are not making much.  And it is to fund other development and documentation activities that will benefit the public in the end.

GlowPon3, 2023-03-31:
I'm just thankful they are putting these together for us, they aren't even that expensive.  Besides, it's not that hard to model something similar yourself.  There's no need for such aggression.

MsnSazabi, 2023-04-02:
> 

Mellow wants 108USD for shipping for 4pcs to me for some reason. This is why I'd make those myself. I'd better tip directly to you.

MirageC79, 2023-04-04:
> > 
> 
> Mellow wants 108USD for shipping for 4pcs to me for some reason. This is why I'd make those myself. I'd better tip directly to you.

Each country has fees and taxes that need to be added to the shipping price.  Mellow is only applying what your government is charging them to get stuff in your country.  

Here is how much shipping is to Canada:

[image]

closed 2023-04-04 by MirageC79

MirageC79, 2023-04-04:
As mentioned in the video, this is the design that was used as the source idea: https://www.thingiverse.com/thing:4781610
WS8 should not be printed, they are too small and will delaminate.  WS12 printable version are posted up here.

Not sure what you mean by "Corrected" open source idea?  Are there issues with the original design?

pasha4ur, 2023-04-04:
> WS8 should not be printed, they are too small and will delaminate.

Or will not delaminate.

pasha4ur, 2023-04-09:
> As mentioned in the video, this is the design that was used as the source idea: https://www.thingiverse.com/thing:4781610 WS8 should not be printed, they are too small and will delaminate. WS12 [...]

https://www.thingiverse.com/thing:4781610

HevORT Wobble Rings for FSU1204 ball screws by[EvoMotors](https://www.thingiverse.com/EvoMotors) is licensed under the [Creative Commons - Attribution - Non-Commercial](https://creativecommons.org/licenses/by-nc/4.0/) license.

Non-Commercial