MirageC79/Interfaces-for-WobbleX-integration #4: RatRig V-Minion

TekuSP, 2023-03-31:
Hey,

When we can expect V-Minion version? I have constant fight with wobbling.

Thank you

MirageC79, 2023-03-31:
I hope the community chips in with these designs, I will try to address as many printers as possible, but the list is growing FAST! 
Meanwhile, have you seen this video: https://youtu.be/c6JmCdovE0U
Many sources of artifact may seem to be Z wobble, but they end up being something else.

junk495, 2023-04-07:
Hello, I did it that way: https://github.com/junk495/WobbleXII-V-Minion/tree/main/RatRig/V-Minion 
Works pretty good so far.
[image]

junk495, 2023-05-17:
Thank you for the feedback and the pictures, makes me happy to see that.

TekuSP, 2023-05-26:
> Just wanted to share, wobble x arrived today, solved all my issues with Z Wobble. Printed with 250mm/s ! The weight of the gantry is more than enough to keep it in place at higher speeds. ✌️ [image] [image]
> 
> [image]
> 
> [image]
> 
> Thank you @junk495 @MirageC79

I have tested it @memphisdj and only result I got was Wobble X flying off while trying to do just HOME Z. Can you provide better pictures? What extruder/hotend combo you have? Because my gantry feels too light for this. Maybe I am doing something wrong.

MirageC79, 2023-05-26:
Oh! wow!  This is awesome!  
@junk495  what a clever idea!  I love how you installed the wobbleX right at the motor coupling!  This Idea should simplify the installation for many printers!

@memphisdj Glad that you like it!  Great print 👍🏼

ZeyHex, 2023-07-02:
I thought it would be cool to use the suggested coupler adapter and just make a split coupler. It turned out like this.
https://www.printables.com/model/518585-ws8-coupling

[image]