MirageC79/Interfaces-for-WobbleX-integration #6: I can't download all step files

OndraRuca, 2023-03-31:
Hi, 
I can't download some STEP file e.g. Ender3 / ZCarriage_WS8.step or MockUp / WobbleX_WS8_PrintedAdapter.step
Some others are OK e.g. Ender3 / TopBearingSupport.step)
Thanks in advance for checking the issue

hapklaar, 2023-03-31:
Have you tried downloading the [entire zip](https://codeload.github.com/MirageC79/Interfaces-for-WobbleX-integration/zip/refs/heads/main)? The step files you mention open fine in Fusion360 on my PC

OndraRuca, 2023-03-31:
That worked, thanks a lot

MirageC79, 2023-04-04:
> Have you tried downloading the [entire zip](https://codeload.github.com/MirageC79/Interfaces-for-WobbleX-integration/zip/refs/heads/main)? The step files you mention open fine in Fusion360 on my PC

Thank you!  For some reason some Steps are interpreted as downloadable by GitHub and others not.  Thank you for the tip.  Closing issue

closed 2023-04-04 by MirageC79