MirageC79/Interfaces-for-WobbleX-integration #7: Ender3 Nut size

leecuk, 2023-04-01:
Hi, I believe the WobbleX_TR8_Adapter Body might have a small issue.

Design has a 16mm screw mount PCD, the brass nut actually has an 18mm PCD.

Thanks
Lee

[image]

MirageC79, 2023-04-04:
Oh!  sweet! thanks for the update.  I'll revise the design.

julian-margara, 2023-04-14:
Hi there,

I modified the PCD to 18mm, I'm not a design expert by any means but maybe my change will help to someone else.

I can't upload the file here but here is the OnShape document from where it's possible to export to any format (stl, step, etc).
https://cad.onshape.com/documents/cc9a77c7250780a055705ba5/w/e54ded23702a3f2bf6409cc8/e/cab1ab567e6af58cdebe1ea9?renderMode=0&uiState=64395ef437952e7f6c7ce812

I'll update my feedback here after testing it (waiting my kit from AliExpress!)

Best

leecuk, 2023-04-14:
Might need a recess for screw to sit in, the nut is tapped and requires the screw to be from above. I’m not sure how much clearance is needed. Also waiting on my kit.

julian-margara, 2023-04-14:
@leecuk why don't put the screw from below? (honest question)

Look how it looks with the 18mm PCD and using the screws (stock ones) from below:
[image]
[image]
[image]

leecuk, 2023-04-14:
Unless you drill out the thread in the nut, once the screw hits the bottom of the nut thread, it won’t pull the two pieces together. I believe it is sometimes called screw bridging or jacking. 

I also can’t say I’m a fan of tapping directly into the plastic, but a possibly a threaded insert might make it better.

leecuk, 2023-04-29:
@julian-margara
My WobbleX arrived today, I'm not sure if I ended up purchasing a clone, but the included rollers were a little shorter than expected. 

The bottom mount fouls the carriage bracket on the ender3, so I've chopped the corner off.

[image]

[image]

I used threaded inserts for the top adapter.

I've uploaded step files to https://github.com/leecuk/WobbleX-Ender3 if anyone wants them.

nevermor113, 2023-05-17:
> Chào bạn,
> 
> Tôi đã sửa đổi PCD thành 18mm, tôi không phải là chuyên gia thiết kế nhưng có lẽ sự thay đổi của tôi sẽ giúp ích cho người khác.
> 
> Tôi không thể tải tệp lên đây nhưng đây là tài liệu OnShape từ đó có thể xuất sang bất kỳ định dạng nào (stl, step, v.v.). https://cad.onshape.com/documents/cc9a77c7250780a055705ba5/w/e54ded23702a3f2bf6409cc8/e/cab1ab567e6af58cdebe1ea9
> 
> Tôi sẽ cập nhật phản hồi của mình tại đây sau khi thử nghiệm (chờ bộ kit của tôi từ AliExpress!)
> 
> Tốt nhất

can you send me the middle part file? I didn't see them post it online

ESOCHI, 2023-11-12:
> Oh! sweet! thanks for the update. I'll revise the design.

Just fyi, mine was still 16 when I downloaded the Ender3 files.

ESOCHI, 2023-11-12:
> @julian-margara My WobbleX arrived today, I'm not sure if I ended up purchasing a clone, but the included rollers were a little shorter than expected.
>
> The bottom mount fouls the carriage bracket on [...]

How did you get the stepper back on? That 3rd bolt for the top plate conflicts with one of the 4 mounting points for the stepper.

I flipped the top mount 180 and that gave me more room.

I'm just offsetting the stepper by 1cm so I've cut the entire middle circle section out, and I'm re-drilling 2 of the holes. I'll then need to extend the bracket to add the other 2 holes which I may do by drilling an aluminum business card out to fit the stepper, then bolt that to the bracket or something.

[Here's a google album with some pics of everything.](https://photos.app.goo.gl/tgNgBSQznW8CyqgE9)

leecuk, 2023-11-13:
I’ve completely shifted the extruder. I’m running a e3d titan extruder, so made an adapter plate that bolts into the original mount. You can see it in my photos.

leecuk, 2024-03-23:
So, it's been just under a year since I installed the WobbleX on the ender, got to say the results are fantastic, so thanks to everyone that put it together.

In terms of the ender3 v2 files, the ones on this repo still aren't currently compatible, but I've got them over at https://github.com/leecuk/WobbleX-Ender3 if anyone want them.

closed 2024-03-23 by leecuk