MirageC79/Interfaces-for-WobbleX-integration #8: Voron Trident MGN12H rear Z

Exerqtor, 2023-04-03:
I've made a MGN12H version of the rear Z carriage for those who want that (both 2 & 3 hole).

I'll just submit the CAD here, forking and creating a pull request for this is too much work lmao.

[image]
[WobbleX-MGN12H_rear_Z.zip](https://github.com/MirageC79/Interfaces-for-WobbleX-integration/files/11135149/WobbleX-MGN12H_rear_Z.zip)

MirageC79, 2023-04-04:
Awesome Thank you!  I'll leave the ticket opened as a reminder to me to incorporate this tomorrow.

MarkusNiewoehner, 2024-02-07:
**Awesome Thank you! I'll leave the ticket opened as a reminder to me to incorporate this tomorrow.**