MirageC79/Interfaces-for-WobbleX-integration #13: Print files of WobbleX #2

pasha4ur, 2023-04-07:
Hello

You closed topic and didn't answer:  https://github.com/MirageC79/Interfaces-for-WobbleX-integration/issues/2

Will you give files for selfmade (WS8 and WS12) or  you are so greedy that you will sell remade free solution?

3d print community wants to know how low you can fell.

pasha4ur, 2023-04-09:
Source of idea for your parts as you said: https://www.thingiverse.com/thing:4781610

HevORT Wobble Rings for FSU1204 ball screws by[EvoMotors](https://www.thingiverse.com/EvoMotors)is licensed under the[Creative Commons - Attribution - **Non-Commercial**](https://creativecommons.org/licenses/by-nc/4.0/) license. https://creativecommons.org/licenses/by-nc/4.0/

pasha4ur, 2023-05-24:
Print version https://www.printables.com/model/439423-wobble-biscuit/comments

Greedy Mirage will not share

MirageC79, 2023-05-24:
Yeah... I am soooo greedy!  all of the ideas me and my team have put forward over the years... Yet this simple side project is what you use to label me as greedy...

Can't see the forest for the trees...

closed 2023-05-24 by MirageC79