MirageC79/Interfaces-for-WobbleX-integration #15: Trident parts

thiagolocatelli, 2023-04-12:
Just received my 3 WobbleX WS8 to install on the Trident. I have a Voron 2.4, very well tuned (printed a lot of Voron Kits with it) and printed all the parts for WobbleX on my 2.4. Having that said, two issues:

The pins I received have length of ~10mm and thickness of ~2.95mm (0.03 more/less - some are even 2.92mm).

1) None of the pins that came with the WS8 stay on any of the slots in any of the printed parts. They slide right out with just gravity. Shouldn't they be a snug fit? It would be difficult to assemble all 3 Z axis having to worry about pins falling off. I could glue them but also believe it should be a snug fit so I can apply so force while pushing them in.

2) The rear Z carriage mount with integrated slots for the pins. When the pins are in place, the little metal ball does not touch the pins, it rests on the printed part since the rear Z mount does not have the recess like other parts have.

[image]

[image]

If you can get me updated STLs I can go ahead and print them again.

MirageC79, 2023-04-12:
Oh wow!  sorry to hear that.  I assume this is a kit from Mellow3D and not one of the many clones that are appearing on AliExpress lately?

I'll measure the pins I have here in my kit.  I'll make a version of the STEP/STL with smaller pin diameter.  I'll make the holes to be 2.95mm instead of 3.1mm.  I'll try to send you a version later tonight.

Sorry for all the trouble and thanks for the feedback.

Note: since you are the first one I know to attempt install on the trident, feedback on the need for top bearing support would greatly be appreciated.

thiagolocatelli, 2023-04-12:
@MirageC79 do not be sorry! it is all good.

I converted all 1.55mm radius to 1.45mm radius and am printing right now. The small adapter it worked perfectly. Going to test the Z plates and then the rear Z.

For the rear Z I did this and if it works I can send you the step files

[image]

thiagolocatelli, 2023-04-13:
@MirageC79 my adjustments worked me for, making the radius 1.45mm. The rear Z kinda worked because with the recess now that allows the bearing ball to sit flush with the pins, we have those two small faces on each end so had to include a nasty brim there. Wont even share picture here because its not pretty but it does work. If you have another solution/suggestion let me.

Exerqtor, 2023-04-14:
I just noticed these issues too and made the exact same changes to my MGN12H version lol, thought I should make an issue about it and saw @thiagolocatelli had allready done so lmao.

Regarding the top bearing support you mention  @MirageC79, are you refering to a potential need for a top support of the leadscrews?

HK-Moebius, 2023-04-15:
I've the issue too.

HK-Moebius, 2023-04-16:
Found a easy workaround for those who not handy with CAD. 

Open step file in text editor.
Use find and replace to exchange all 1.55 to 1.45.
Save.
Use Fusion360 or other tool to export step to stl.
Prusa Slicer also can handle step files.

Happy printing :)

thiagolocatelli, 2023-04-17:
@HK-Moebius changing the radius on the step file works for the front plates and adapters but we still need the recessed feature on the rear Z so the metal ball sits flush with the pins and not the printed part.

pila81, 2023-04-25:
In addition to the missing feature mentioned above, the recessed nut hole is missing the bridges allowing for support-free printing:
[image]

On a side note: my pins are 2.99 mm and found a slot radius of 1.50 mm worked best for me.

Exerqtor, 2023-04-26:
> I just ran into the issue with the lack of a recess in the rear mount and come here to report it. On the other hand, all the pins fit fine on my printed parts.

I just released a fixed[ MGN9H version](https://github.com/Exerqtor/Voron/tree/main/Mods/vt_rear_wobblex_carriers) as well

MirageC79, 2023-04-27:
Sorry guys, been pulled away for past couple of weeks due to personal stuff.  Will be looking at fixing this over the weekend.
Thank you for the input!  and @Exerqtor great work! May I include your parts in this repository?

Cheers!

Exerqtor, 2023-04-27:
> Sorry guys, been pulled away for past couple of weeks due to personal stuff. Will be looking at fixing this over the weekend. Thank you for the input! and @Exerqtor great work! May I include your parts in this repository?
> 
> Cheers!

Hope you're doing well!

Yeah ofc, do as you like with them 😊

claudermilk, 2023-04-28:
> > Sorry guys, been pulled away for past couple of weeks due to personal stuff. Will be looking at fixing this over the weekend. Thank you for the input! and @Exerqtor great work! May I include your parts in this repository?
> > Cheers!
> 
> Hope you're doing well!
> 
> Yeah ofc, do as you like with them 😊

I actually just found your part on Printables. I just posted my make over there--it works great. I now have WobbleX installed with your updated model. No longer does my Trident howl while running down the Z travel--this mod is clearly working.

Exerqtor, 2023-05-01:
I finally got around to print this (and the front ones) myself, and i had to go with Ø 3.0mm for the pins just FYI.

slashbeast, 2024-05-09:
Thanks @Exerqtor, I am using your remix for a few months and it works great for me. Would be great if those in-repo Trident parts would be fixed, because the rear without the channel to roll ball in is just quite literally unusable.