MirageC79/Interfaces-for-WobbleX-integration #16: Parts for Sidewinder X1

acheljay, 2023-04-18:
Hi, could it be possible to have parts for the sidewinder X1, certainly one of the most wobbly printer I know ? Many thks in advance

washo4evr, 2023-04-21:
would X1 parts fit the X2?

acheljay, 2023-04-21:
I don't think cause the X2 have a new anti wobble system with spring tension for the wheels

gt3rs40, 2023-04-22:
X1 parts would be awesome

Caos-Maker89, 2023-05-16:
waiting for x1 parts :)

acheljay, 2023-07-25:
yup always interrested cause I did'nt find any good one for the X1

gt3rs40, 2023-07-25:
Still interested :)

washo4evr, 2023-07-26:
can this be ported to the X2? How different are the 2 models?
I just received my wobbleX yesterday but not sure how to install on my X2

fkahhaleh, 2023-07-28:
Alright
So Amazon didn't delivery the parts today as planned. Instead of keeping you guys from potentially trying it out.
Here is a link to [Thingiverse](https://www.thingiverse.com/thing:6144103).

It is heavily a WIP/concept that I have not tried.  Keep that in mind :)

Feedback is welcome of course!

fkahhaleh, 2023-07-29:
Perhaps a quick update on my end.
I was surprised that my Sidewinder X1 (v4) already had a similar mechanism to address Z-banding. (basically a blue cube with the brass nut inside with some degree of freedom for movement)
But I did install one side and I noticed that when there is z-axis movement, the rotation actually gets the Wabble-X moving too much that it actually almost jumps the metal ball bearings. Perhaps this is similar to the weight distribution issue mentioned in the wiki.

I ended up replacing the original parts for the X1 because the above design I provided did not work!

ApolloF, 2023-08-24:
You can just use the mockup parts for the x1.