MirageC79/Interfaces-for-WobbleX-integration #18: WobbleX ball size

woodwaker, 2023-05-11:
I ordered 6 WS8kits from the Mellow store and just installed the first set on my Rat Rig 3.1.  During the installation I lost one of the ball bearings, I can not find the specs for them,  could you include the sizes for each kit?  I also left a message for Mellow, it might be a good idea to either include an extra ball or have them as an option.

woodwaker, 2023-05-11:
Thanks for making me feel better, not being the only one that had lost a ball.  I have an assortment on order from Amazon, and will update if I have success.

woodwaker, 2023-05-12:
These are 4mm ball bearings.  I was in a hurry so ordered an assortment from Amazon
https://www.amazon.com/gp/product/B08KTBM74G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1
I would suggest that your order some from AliExpress the same time as you order the WobbleX

MirageC79, 2023-05-18:
Sorry to hear that, yes they are 4mm Steal balls.  I have asked Mellow to put extra in the kits.
Thanks for reporting this.

closed 2023-05-18 by MirageC79

woodwaker, 2023-05-18:
Thanks for the reply!
I am sure it was my fault, but I could not figure out a good way to get the WobbleX stack installed in each Z axis arm without balls dropping out.
I ordered an assortment that included 50 4mm ball and I think both of my Rat Rigs are now running well.My friend Tim of DWIT3D Canada is going to install on his RatRig in the next week or so, I ordered the same set for him so he will be prepared.I think he will do a live steam of the install.  Here is his channel   ***@***.***/streams
I see a couple of problems, the magnets don't seem to be powerful enough for the stainless steel balls and they are not constrained. 
I putting everything back into the Z axis printed part it is easy to slightly move the alignment and the balls find freedom.

dave wilson 816 792 1060