MirageC79/Interfaces-for-WobbleX-integration #40: Ender 3 ball jumps the pin

moff3tt, 2024-09-20:
Hello,

I installed the wobblex on my old ender 3 pro and I find that one of the balls tends to jump over the pins and ride the outside of one instead of in the center.

I'm not sure what is causing it. I have installed everything as close to the instructions as possible and my z is so well aligned it will fall on its own with the stepper off.

(Note I had to use a little bit of hot glue to keep the pins in the printed parts since mine came out a tad loose)

[image]

moff3tt, 2024-09-21:
further analysis of the issue has provided the root cause. When my bl touch fails to probe the whole system goes slack which allows the WobbleX to come loose since it falls a little more than the x gantry then when I reset the printer and the z axis goes back up the ball is misaligned. Any guidance on how to mitigate this issue so manual intervention is not required is appreciated.

leecuk, 2024-10-20:
Maybe the same approach as when using 2 lead screws, some people use bands etc to pull them together, although sounds like your real problem is the bl touch, is it genuine or a clone?, I've only used a clone and it was junk, I've got a genuine creality probe and it works flawlessly.