MirageC79/Interfaces-for-WobbleX-integration #41: Formbot Marathon - Interface Plates Pinching/Clapping

phiferch, 2025-01-31:
I recently purchased a Formbot Marathon printer, which has 3 ball screws, with the WobbleX mechanism attached to all three. I noticed that when I lowered the bed, sometimes the interface plates of the WobbleX would sorta bind up and pinch, making this unsettling clapping noise. Is this the same issue as the bed needing to be heavy enough to keep the plates together? Any assistance would be greatly appreciated. Video for reference.

https://www.youtube.com/watch?v=QI-7QPU0SkM

leecuk, 2025-02-02:
I'm not sure it is, I can't actually see any ball bearings between the needle bearings, are they missing?

Did you do the install?

@phiferch

leecuk, 2025-02-04:
Assuming the top plate over the wobblex is attached correctly with no issues, then it might be the cause you suggested. 

Does it work ok if you apply downward pressure to the build plate when lowering?

I’m not familiar with the printer, but a quick Google came back with someone on Reddit having separation issues and were using springs to pull the plates together. 

https://www.reddit.com/r/FixMyPrint/comments/190a40r/thoughts_on_my_wobblex_adapter/?rdt=42573

Something like rubber o-rings may work, but not sure how it would affect the performance. 

You may find the effort isn’t worth the reward, but I don’t know how badly your printer performed beforehand.

My only experience of the wobblex is with the ender3, but it really does work on that.